prunkdump/arduino-variometer #1: Thanks

hw999, 2016-12-30:
Hi prunkdump

thanks for publishing your vario solution. It was very helpful for my own project - especially vertaccel and kalmanvert. Performs really nice.

I plan to do make an enhanced version where speed also goes into the equation, known as TECS.
There are already calculations available. But the ones I know don't make a fusion of vertaccel, pressure and speed. 
I would like to have this in a Kalman filter - for the time being I lack the knowledge how to do that.

Do you maybe have some hints ?

regards Andi

prunkdump, 2017-01-01:
Hi Andi !

I'am happy that you find my Kalman code helpful ! It was the reason that motivate me starting this project.

I'am interested about your project and how your embed my code. I need as much feedbacks as possible to improve my libraries. Can you send me some ?

This week a have a lot a work. But I can send you a three dimensional Kalman code. I just need a little time to writing it well. But be careful that it will be more computer intensive.

Last question. Do you use your MPU9250 with a 3.5V arduino ? I have some trouble using my Kalman filter with only 3.5V (Atemega328P + MPU9250 + ms5611). But maybe it's an hardware problem.

Maybe you can email me on :   

prunkdump at gmail dot com

So I can send you private messages.

Regards, Baptiste.

hw999, 2017-01-17:
Hi Baptiste

I was late with an answer because I didn't notice your reply. Did receive my email ?

kostashellas, 2017-05-03:
Hi Baptiste,
Do you consider changing AVR to something like SAMD21, more performance, memory.
I have a little project that I'm working on the last 6 months and I've reached the Arduino micro limits that's why I've

kostashellas, 2017-05-03:
I'm migrating the entire project to MicroZero that has also SD build-in.
I'm integrating to it your 5611 code and adding the GPS functionality.  All information send to the user   via voice messages thru the headsets.

Best, Kostas

prunkdump, 2017-05-03:
Hi Kostas ! 

Currently I try to finish my code for Atmega328P as it is the most common hardware used by Arduino hobbyist.

But I can help you if you want. You just want an ms5611 library for cortex M0 ?

Regards, Baptiste.