prunkdump/arduino-variometer #2: Réglage du son

jpg63, 2017-04-30:
Le son est un peu fort et il serait intéressant de le régler

Priorité : Indispensable

solution :

Pour le volume c'est effectivement possible de le régler avec le code. Je comptais faire la modif. Il faut remplacer tous les toneAC(freq) en toneAC(freq, VOLUME) avec un volume entre 1 et 10. Mais encore une fois il faut que j'explique comment mettre à jour le code.

van-hurlu, 2017-04-30:
Très facile mettre le code à jour, je l'ai fait pour arrivé à faire fonctionner le baro
si j'ai pigé ma manip
il suffit d'ajouter ,5 juste après la valeur freq dans le code
ou mieux de créer un int vol
et de mettre ,vol après la valeur freq dans le code

prunkdump, 2017-04-30:
Continuez la discution sur variometer-kits.

closed 2017-04-30 by prunkdump