prunkdump/arduino-variometer #6: XC Track implementation

YgorTN, 2017-06-12:
Hi Prunk Dump,
I have done several flights with this setup: MS5611, LCD 5110, Speaker and Bluetooth HC-05 using your code (Commits on May 23, 2017 - send vario sentences without GPS). I used it also with XC Soar Android app.

I'm more than satisfied with its performance. Thank you for sharing your idea. 

Another, probably more suitable app for paragliders available on Android store is XC Track.
The only problem is, that this app does not support lxnav bluetooth sentence protocol. 
This app support only these comercial varios e.g.: FlyNet, BlueFly Vario etc.
http://xctrack.org/External_Sensors.html 

Do you think that there is an option to build in some extra protocol (e.g. same as BlueFly Vario), that can user choose in code to make your vario functioning with XC Track?
 
Thank you. 

Regards, Igor

prunkdump, 2017-06-13:
Hi Igor !

Have you tried the XCtrack setup in real condition ? Because on the changelog :

http://xctrack.org/Change_Log.html

there is  :
---------------------------
2016-05-26 0.5.1.2 (development version)

    XCTracer support (baro data only, protocols: XCTRACER, LK8EX1, LXWP0)
---------------------------

So XCtrack should already support LxNav sentences.

Regards,

Baptiste.

YgorTN, 2017-06-14:
Hi Baptiste,

Sure, I've tried the XC Track setup in real condition. I can only connect my HC-05 BT module to the app, but I can't get any measured pressure data in 'Calibration'. 

I've tried to switch 'Atmospheric pressure sensor' into option 'Use internal barometer' in the app settings (My phone LG G3 has internal pressure sensor) and in 'Calibration' I see mearured data in real time and vario is working  (not precisely).

In addition, there is no option to choose a specific device or protocol as it is possible in XC Soar app.

I've also tried to use antoher Android smartphone (without internal sensor), but it aslo hasn't been working. 

Regards,

Igor

prunkdump, 2017-06-14:
Hi Igor !

It's seems that XCTrack need a higher frequency for the LxNav sentences. If you still have the same version of the code. Try to change : 

#define VARIOMETER_SENTENCE_DELAY 2000

in

#define VARIOMETER_SENTENCE_DELAY 500

Baptiste.

YgorTN, 2017-06-15:
Hi Baptiste,

I've already tried to experiment with sentence frequency in range 100 ms - 2000 ms including your suggestion 500 ms. Unfortunately, without any success. 

Today I have found this: https://www.xctracer.com/en/faq/?oid=1856&lang=en

The XCTracer support these protocols:
#supported protocols are None, XCTRACER, LK8EX1 or LXWP0

XCTrack (Android): BLE string XCTRACER, forwardGPSSentences=no

Whereas, XCSoar (Android, Kobo): BLE string LXWP0, forwardGPSSentences=yes, choose LXNAV as driver in XCSoar

So this is probably the reason, why current setup works only with XCSoar app. 

Igor

prunkdump, 2017-06-17:
I don't think that this is the origin of the problem. It is just that XCTrack understand the "XCTRACER" sentences. So it is possible to send them directly.

Few months ago someone used this code with XCTrack with small adaptations. So it is possible. Sadly I don't have a device to try XCTrack myself. I will try to recontact him to know how he setup the code.

Baptiste.

YgorTN, 2017-06-19:
Hi Baptiste,
That would be perfect, thank you.

Igor

prunkdump, 2017-08-22:
Hi Igor.

The code now support XCTrack with the LK8000 NMEA sentences !

You have just to change in librairies/VarioSettings/VarioSettings.h :

#define VARIOMETER_SENT_LXNAV_SENTENCE
to
#define VARIOMETER_SENT_LK8000_SENTENCE

Thank you very much for the bug report !

Baptiste.

closed 2017-08-22 by prunkdump