prunkdump/arduino-variometer #7: BMP280

Hasi123, 2017-06-19:
Hi,

Thanks for sharing your project. I have a suggestion for you: You could try to implement the Bosch BMP280 barometer sensor. It is much cheaper, than the MS5611 and has almost the same performance.

prunkdump, 2017-06-21:
Yes you're right, this need to be added to the project !

Even if I think that the BMP280 does not really have almost the same performance than the ms5611.

I contact you when I have found the time to work on it.

Thanks !

Baptiste.

Hasi123, 2017-08-26:
I made a quick and dirty adaptation for the BMP280 using Adafruit's library. It works, but is probaby very inefficient. https://github.com/Hasi123/arduino-variometer

Hasi123, 2017-09-03:
I changed Adafruit's library to use your fastwire communication.

prunkdump, 2017-09-13:
Hello Hasi !

I'am really sorry ! I accidentally removed watching on this issue so GitHub don't send me emails when you put your comments.

I checked you work and it seems good ! Have you tried in flight ?

The only problem is that you feed the Kalman filter each time you get a new value from the accelerometer, regardless of if you have a new value from the barometer.

When I have some times, I will try to adapt you code.

Thanks you for the help !

Hasi123, 2017-09-13:
Since I have a broken leg, I cannot test it in flight yet. As soon as I`m able, I will test it.

Hasi123, 2017-10-27:
hi

Hasi123, 2017-10-27:
Finally I can fly again and was able to test the vario. It worked quite well even with my faulty Code. It would be nice to see it implemented with your routine with the interrupts.

Hasi123, 2017-10-28:
So new update: I read through the BMP280 datasheet and the sensor reads the pressure at a stable frequency internally in normal mode, so there is no need to use timers. I changed the BMP280's settings a bit to better suite our application.

Hasi123, 2017-11-20:
Finally I could test the vario in thermally conditions and it works like a charm! I did a 30km flight in the middle of november. It is very accurate and does not suffer from sudden pressure changes due to an increase or decrease in speed, that I sometimes had with my old homemade vario (only barometer with a low pass filter).

prunkdump, 2017-11-22:
That's great !!!

I have just ordered a BMP280 board to improve the code with you.

The BMP280 have multiple modes. The only drawback of using the "continous" mode is that it consume more energy.

When I have the time to work on It I will sent to you a version of the code with timers interrupts. So you can compare the two versions.

I'm also very interested by you 1.8V setup !

Good job !

Baptiste.

m00nsp3ll, 2019-04-05:
bmp280 have also temperature sensor on it, it could be useful to implement temperature data into vario if bmp280 usage is possible

Hasi123, 2019-04-15:
The temperature data from the barometer is actually used to compensate altitude readings with both the ms5611 and the BMP280.

It would be interesting to test the BMP388, because it has even more resolution and accuracy than older sensors...

prunkdump, 2019-04-27:
The code now support the BMP280 barometer.

Thanks for the report.

closed 2019-04-27 by prunkdump