prunkdump/arduino-variometer #8: Vertical speed but no change in altitude

sbishop61, 2017-07-29:
HI,\ a great project and contribution.

I have a breadboard set up with Pro Mini,  display and MPU9250 & MS5611.

When it is not moving on the desk, the altitude does not change but there is always a vertical speed reported; typically +0.3. After 5 minutes the altitude is still the same and the there has always been a vertical speed.

I thought that the vertical acceleration from the DMP was used in a kalman filter applied to the pressure (altitude) reading. I don't understand how if the altitude is not changing.

Any help appreciated.

Regards,\Stephen

prunkdump, 2017-07-31:
Hi Stephen !

I very happy you like my project.

The kalman filter is not a "long term" filtering algorithm. It use just, each time, the previous step and the new value. Search for "kalman filter" on the web for an introducing article.

Now what's happen in you case. The DMP is bad calibrated so it sent constantly an upward force. The barometer sent nearly always the same altitude.

-> on one step the kalman filter receive 235m and an upward force. 
-> as the upward force must increase the vertical speed it assume a +0.3 vertical speed.
-> the next step, the kalman filter receive again 235m and the same upward force. This is not coherent with the privious measure as the altitude have not encreased. So the kalman filter try to make the best assumption.

There is an upward force so the vertical speed must be encrease. But the altitude remain the same so vertical speed must be 0. So il choose 0.3 again.

Otherwise you need to calibrate your gps. Use the "calibration" or the "calibration_nointerractive" sketch.

Don't hesitate if you have questions.

Regards.

Baptiste.

sbishop61, 2017-07-31:
Baptiste,

I assumed that the kalman was used to smooth/filter the pressure readings and that the vertical velocity was calculated from the change in the compensated altitude. In that scenario a constant acceleration might effect the filter output but if the altitude was constant the vertical velocity would always be zero.

I have tried to use the calibration routines but it is hard when on a breadboard. When I ran the calibration it always failed saying BAD measurement.

regards,

Stephen

prunkdump, 2017-08-02:
Hi Stephen !

Are you sure that you always obtain bad measurement ? Normally you should obtain at least one good measure for each orientation the first time. The second time you get "bad" measure if the acceleroleter is less stabilized than the fist time. Do not touch the breadboard during measurement.

Another tips. Take a box and fix the breadboard inside with tape. So you can use the calibration_nointeractive sketch following this procedure :

https://www.youtube.com/watch?v=fc0vIsZJvsw

Good luck.

Baptiste.

prunkdump, 2017-08-22:
Hi Stephen !

It's seems that you find the problem.

I close the issue.

Baptiste.

closed 2017-08-22 by prunkdump