prunkdump/arduino-variometer #10: arduino-variometer to android-variometer?

lshachar, 2017-08-19:
I watched the videos on youtube - and your project seems quite impressive!
on my galaxy s4 smartphone there is a bmp180 barometric sensor, and obviously a 9dof sensor.

It appears that none of the currently available android vario apps integrate the 9dof sensor..
Do you think a port to android may be feasible?
I'm trying to figure out if I can do it myself, but I'm not a very good programmer (I'm experienced with arduino, but not at all with android... I might just end up building your project, will take me less time than trying to port the code to android!)
I'll be happy to hear your thoughts about it!

lshachar, 2017-08-19:
hmmm...
[use sensor fusion to improve vario responsiveness ](https://github.com/msarto/AVario/issues/2)

prunkdump, 2017-08-22:
I Ishachar.

Excuse me for this late reply !

Sadly I don't have any practice in Android programming. But I know that :
-> It's possible to use C++ with Android with Android NDK so you can port some part of this code.
-> It's possible to access the 9dof and bmp180 data with Android libraries.

So porting this project to android seems to be possible ! If I find some more info I will send to you the links.

Thanks for the idea !

Baptiste.