prunkdump/arduino-variometer #11: vario + bluetooth+ gps = no sentences

lshachar, 2017-10-19:
Hello prunkdump!
I have built a vario using your code. A+++ stuff! thank you!!
I now got a GPS and Bluetooth installed in my vario, and I think I'm onto a bug.

in variometer.ino:323 `#else //!HAVE_GPS`
When there is GPS installed, but GPS didn't get a fix, no sentences will be sent.
I believe that this is unintentional behavior; in this case the vario should send pressure and vario data, neglecting calibrated altitude (since it cannot be calibrated - no GPS). 

once GPS gets a fix, calibrated altitude should be sent.
I think that excluding this code (with line 323) when there is GPS and Bluetooth is the problem.
Do you agree?

prunkdump, 2017-10-20:
Hi Ishachar.

No there is no problem. I explain.

When the variometer sent pressure/vario sentences it must be carefull to not sending the sentences at the same time than GPS. Because they use the same serial port. So synchronization is needed between the GPS and the variometer.

So : 
-> When the GPS is installed. Pressure/Vario sentences are sent just after the last GPS sentence. It's the GPS that give the rate of the Pressure/Vario Sentences. Search for "lastSentence = true;" in the code. And the GPS sent the sentences even if it didn't get a fix.

-> When there is no GPS installed. The variometer is free to sent the sentences when it want. So there is just a delay between sending. And the "last sentence" code is disabled. This delay can be set in "VarioSettings.h".

Regards,

Baptiste.

jpg63, 2017-10-20:
Salut Baptiste,

tu as du mélanger mon mail avec celui de Ishachar

Pense à lui renvoyer ta réponse

A+

kostashellas, 2017-10-20:
Hi Baptiste,
That means the vario refresh rate, in case of  GPS, = GPS refresh rate thus
20Hz?

Regards Kostas

prunkdump, 2017-10-21:
Yes ! I known that this must be improved !

GPS often use a refresh rate of 0.5Htz. One time every two seconds. And for the Pressure/Vario sentences it's a little bit slow.

Thanks for the report !

defvs, 2018-01-08:
How about using SoftwareSerial, by changing bluetooth pins and completely separating the two ? I know there is a rom space issue, but with a little tweaking it could be almost invisible.

**Français**
Pourquoi on utiliserais pas SoftwareSerial ? On change les pins du bluetooth et on sépare complètement les deux ? Apparement il y a peu de place dans la rom avec le programme mais avec un peu de tweak on peut réussir à négliger l'impact sur le processeur de l'arduino

If you need more help just ask me; si tu veux plus d'infos demande moi

prunkdump, 2018-01-08:
Hi Defvs !

There is two problems with SoftwareSerial : 

-> Obviously, SoftwareSerial consume more ROM as the Serial protocol internals are implemented by software and not by hardware.

-> SoftwareSerial (like Serial) use two ring buffers, one for RX and one for TX. The SerialNmea library use only one common buffer and filter the input (RX) characters to keep only the needed NMEA sentences.

Thus we save ROM with optimized code but also SRAM with the only one (small) ring buffer.

But you're right. The drawback is that the code is not portable and only the hardware serial pins can be used.

But if you want, software serial can be easily used inside SerialNmea to use some other pins.

Thanks for the suggestion !

Baptiste.

defvs, 2018-01-08:
Hey ! I'll try and make a port ;)

lshachar, 2019-09-11:
I went back to enabling the GPS module on my vario (I decided to disable it back in October 2017..)
And I found the cause of my original issues.
I am using a chinese arduino nano, which has a CH340 chip instead of an FTDI chip. (for USB to serial).
My GPS module, NEO-6m, wasn't able to sink the RX1 pin (on the arduino) to LOW. So even though the GPS module was in fact sending NMEA sentences at 1HZ, the cheap arduino nano did not see that data on its RX1 line.
(I figured that out by using the software serial library, and connecting the GPS TX pin to arduino on a different pin (pin 2). which worked fine.)
Hence, the CH340 is doing something funny - maybe it has a small pull up resistor when it is not trying to send data to the arduino.

In order to solve the problem: I added a PNP transistor (2N4403 I had laying around) between the gps module (TX) and the arduino (RX) that sinks the RX pin to Ground (when the gate goes low - which is connected to the GPS TX pin).
once the arduino received the data on the RX line, I was able to `#define HAVE_GPS` in `variosettings.h` and the info is sent properly on the arduino's TX line (both the gps' NMEA sentences and the vario's LK8000 sentences.

I also learned something else...:
while sending data to the arduino through the serial interface (through UART, which ends up on the arduino's RX1 pin) and at the same time having the GPS module trying to send data (because that's what it does every second)  - I had something goes up in smoke! and that cheap arduino board died. well, probably only the CH340 died as that board won't connect through USB anymore. but otherwise it still works.
I assume that the CH340 tried to drain too much current (both the arduino RX1 and the GPS module should have pullup resistors; once the CH340 was trying to send data it tried to sink both these sources to ground (before the NPN transistor was added) ).
To play it on the safe side I added a SPDT switch on the RX1 pin, and now I disconnect the NPN transistor every time before I upload any code.
(I believe adding a resistor on the transistor's gate, should also keep this from happening again, but I'm not 100% sure that the CH340 burned, and that it happened for the reason I think it did. So I'm playing it safe).

This wouldn't have happened if I used a genuine arduino board.
I wanted to get this information out there in case it help somebody one day.

[image]

closed 2020-04-07 by lshachar