prunkdump/arduino-variometer #12: SETTINGS.TXT

pbwild, 2017-11-23:
Hi!
Could you post an example of the SETTINGS.TXT file for the M0 branch?
Man thanks!
Peter

jpg63, 2017-11-23:
https://github.com/prunkdump/arduino-variometer/tree/jpg63-M0-version/libraries/VarioSettings

[https://avatars2.githubusercontent.com/u/4787331?s=400&v=4]<https://github.com/prunkdump/arduino-variometer/tree/jpg63-M0-version/libraries/VarioSettings>

prunkdump/arduino-variometer<https://github.com/prunkdump/arduino-variometer/tree/jpg63-M0-version/libraries/VarioSettings>
github.com
arduino-variometer - Arduino based variometer with accelerometer, screen, gps and sdcard code

Hi, the file SETTINGS.TXT is present in the directory variosetting This file is placed in the root of the SD card

jpg63

pbwild, 2017-11-25:
Ah yes - found it!
Many thanks!