prunkdump/arduino-variometer #13: M0 Accelerometer Calibration

pbwild, 2017-11-30:
Hi,
do you have a tip on how to calibrate the MPU9250 Accelerometer on the M0 branch. I have got the sketch up and running on an Arduino Zero without an SD Card (btw. in the code VarioSettings GnuSettings; needs to be moved outside the #ifdef HAVE_SDCARD for this to work), but I'm not getting very far with the calibration sketches - I don't quite understand what you are doing with Fastwire.
Many thanks,
Peter

jpg63, 2017-11-30:
Hi,

The calibration of the MPU9250 is not yet implemented in this code, it will come later. For the Gnusettings library it is imperative to have an SD card, the MKRZero has one of them. The setting is stored in SETTINGS.TXT at the root of the SD card. Without SD card the code does not work, it will be possible to make changes later but I do not see the interest.

jpg63

pbwild, 2017-12-03:
OK thanks, looking forward on seeing how you calibrate things on the zero platfom.
Cheers,
Peter

DonCann, 2017-12-08:
I adapted the calibration code t from the samd21 branch to make it work with the zero, doesn't work though, I keep getting this message for any orientation, for instance pointing upward:

accel=(-0.01402, 0.00588, 0.11548)
Standard deviation = 0.62822
BAD measure, not recorded, possible problems are :
-> ambiguous orientation
-> orientation already done with lower standard deviation

Any idea why the orientation is ambiguous?

prunkdump, 2017-12-09:
Hello !

This is not the problem. "Standard deviation" is just the measure of acceleration data "quality". For each orientations, the calibrator record the measure only if the Standard deviation is lower than the previous measure.

The problem is that your accel vector give strange data :
accel=(-0.01402, 0.00588, 0.11548)

Normally, the norm of the vector must be around 1 (due to gravity) with is not the case here. Two possibilities :
-> You MPU9250 does not work well.
-> There is a problem in the Samd21 code.

I have tested the vertaccel_library on Samd21 and it work well. I think than Jpg63 have tested the calibration code without midification.

Can you confirm Jpg63 ?

Don't hesitate if you have questions or if you find something.

Baptiste.

jpg63, 2017-12-09:
Hello The code is that of the original accelcalibrator library and the last MPU library that was tested with the samd21. I did not make any changes other than saving the calibration in memory and on the SD card jpg63

DonCann, 2017-12-11:
Thank you both for your answers, I’ll test with another mpu and let you know.
Cheers

pbwild, 2017-12-30:
Hi,
I've just got everything up and running on an MKR0, but I'm still unclear on how to calibrate. 
Is what DonCann is trying the correct way to proceed?
@DonCann - care to share any progress with your code modification?
Many thanks,
Peter

prunkdump, 2018-01-08:
Hi PbWild !

Which branch do you use ?
-> If you use the master-Samd21 branch the accelerometer calibration is not implemented yet. 
-> If you use the jpg63-M0-version branch, the calibration procedure is automatically launched at power On If not already done. And you need to follow there steps : 

https://www.youtube.com/watch?v=fc0vIsZJvsw

DonCann, 2018-01-17:
I tested my code with two different mpu-9250 and in both cases got values in the 0.01-0.10 range as I mentionned above.
Will report when I find where the problem.

prunkdump, 2018-01-20:
I'am really sorry ! You're right. There is a bug in the calibrator that can cause overflow with high speed micro controllers.

The bug is corrected in the master branch.

I will send to you a Samd21 version soon.

Sorry again. The problem was not your mpu9250 ...

prunkdump, 2018-01-20:
The problem is now corrected. The "calibration.ino" sketch run and display correct values. But the calibration correction coefficient are still not recorded.

Jpg63 you need to port the last correction to your branch ;)

jpg63, 2018-01-20:
I have already updated my pro-mini branch. For the M0 version, the libraries are very different, I have to look

pbwild, 2018-02-16:
Hi, any updates on calibration for the M0 branch? Where exactly should the coefficients be saved - eeprom or flash.txt?
Thanks,
Peter

jpg63, 2018-02-16:
Hi,

The parameters are stored in 2. The eeprom of the M0 is removed at each program update. Parameters are taken first on the eeprom and if they are zero, on the SDcard in a specific file

jean-philippe

kostashellas, 2018-02-16:
Hi JP,

SAMD21 has no EEPROM if not mistaken, do you mean from RAM to Flash?

Flash Memory 256 KB
Flash Memory for Bootloader 8 KB
SRAM 32 KB
EEPROM no
Best, Kostas

prunkdump/arduino-variometer/issues/13#issuecomment-366201148>, or mute the
auth/AY3WRszg8wmlkg2Hxbsz094-CL6ldyQkks5tVVsJgaJpZM4QwVmz>.

jpg63, 2018-02-16:
actually the M0 has no eeprom but it is possible to use the program memory, with the default of seeing the parameters erased at each update