prunkdump/arduino-variometer #15: Cant compille ino.

damirm57, 2018-01-26:
Hi I can  not compile variometar ino. mesage is:
Arduino: 1.6.12 (Windows 7), TD: 1.35, Board: "Arduino Pro or Pro Mini, ATmega328 (5V, 16 MHz)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\damir\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\damir\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\damir\Documents\Arduino\libraries -fqbn=arduino:avr:pro:cpu=16MHzatmega328 -ide-version=10612 -build-path C:\Users\damir\AppData\Local\Temp\arduino_build_833406 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\damir\Desktop\arduino-variometer-master (1)\arduino-variometer-master\variometer\variometer.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\damir\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\damir\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\damir\Documents\Arduino\libraries -fqbn=arduino:avr:pro:cpu=16MHzatmega328 -ide-version=10612 -build-path C:\Users\damir\AppData\Local\Temp\arduino_build_833406 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\damir\Desktop\arduino-variometer-master (1)\arduino-variometer-master\variometer\variometer.ino
Using board 'pro' from platform in folder: C:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20
Using core 'arduino' from platform in folder: C:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" "C:\Users\damir\AppData\Local\Temp\arduino_build_833406\sketch\variometer.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src" "C:\Users\damir\AppData\Local\Temp\arduino_build_833406\sketch\variometer.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src" "-IC:\Users\damir\Documents\Arduino\libraries\VarioSettings" "C:\Users\damir\AppData\Local\Temp\arduino_build_833406\sketch\variometer.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src" "-IC:\Users\damir\Documents\Arduino\libraries\VarioSettings" "-IC:\Users\damir\Documents\Arduino\libraries\I2CDev_ms5611" "C:\Users\damir\AppData\Local\Temp\arduino_build_833406\sketch\variometer.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src" "-IC:\Users\damir\Documents\Arduino\libraries\VarioSettings" "-IC:\Users\damir\Documents\Arduino\libraries\I2CDev_ms5611" "-IC:\Users\damir\Documents\Arduino\libraries\ms5611" "C:\Users\damir\AppData\Local\Temp\arduino_build_833406\sketch\variometer.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src" "-IC:\Users\damir\Documents\Arduino\libraries\VarioSettings" "-IC:\Users\damir\Documents\Arduino\libraries\I2CDev_ms5611" "-IC:\Users\damir\Documents\Arduino\libraries\ms5611" "-IC:\Users\damir\Documents\Arduino\libraries\vertaccel" "C:\Users\damir\AppData\Local\Temp\arduino_build_833406\sketch\variometer.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src" "-IC:\Users\damir\Documents\Arduino\libraries\VarioSettings" "-IC:\Users\damir\Documents\Arduino\libraries\I2CDev_ms5611" "-IC:\Users\damir\Documents\Arduino\libraries\ms5611" "-IC:\Users\damir\Documents\Arduino\libraries\vertaccel" "-IC:\Users\damir\Documents\Arduino\libraries\LightInvensense" "C:\Users\damir\AppData\Local\Temp\arduino_build_833406\sketch\variometer.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" "-IC:\Users\damir\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src" "-IC:\Users\damir\Documents\Arduino\libraries\VarioSettings" "-IC:\Users [... cut ...]
Using library beeper in folder: C:\Users\damir\Documents\Arduino\libraries\beeper (legacy)
Using library LPtoneAC in folder: C:\Users\damir\Documents\Arduino\libraries\LPtoneAC (legacy)
Using library varioscreen in folder: C:\Users\damir\Documents\Arduino\libraries\varioscreen (legacy)
Using library digit in folder: C:\Users\damir\Documents\Arduino\libraries\digit (legacy)
Using library NmeaParser in folder: C:\Users\damir\Documents\Arduino\libraries\NmeaParser (legacy)
exit status 1
'NmeaParser' does not name a type

prunkdump, 2018-01-26:
Hi !

You can't put the content of the arduino-variometer-master where you want. It must be put inside the "Arduino" folder.

So.
1) Empty your Arduino folder by putting all the files in one other location.
2) Put the content of the arduino-variometer-master folder inside your Arduino folder

Be carefull. Don't put the arduino-variometer-master folder inside the Arduino folder but just it's content.

Regards,

Baptiste.

damirm57, 2018-01-27:
Thank you, thank you!!!! Upload is done!

damirm57, 2018-01-27:
Ok, one more question. Is it posibile to use ms5611 sensor on separate board(gy-63) and another board with mpu-9250?  Or I have to bye new one where is both sensor on same board? According to this sketch.

prunkdump, 2018-01-29:
(I have sent to you a reply by mail)

Hi Damir !

Yes you can put as number of I2C boards as you want on the A4 and A5
pins. But sometimes you need to add two pull-up resistors. So :

-> Try first with only the two boards and check if it works.
-> If it don't work, add also a 10k or 4,7k resistor between A5 and
3.3V and a 10k or 4,7k between A4 and 3.3V

Don't hesitate if you have more questions !

Baptiste.