prunkdump/arduino-variometer #16: No conection with gps neo m8n

damirm57, 2018-01-29:
Hi, one more question. I have gps ublox neo m8n conected to vario, but i can not see any update only stay km/h without 0 or any number, Do I need some extra settings with ublox centar before conect to vario? On ublox centar gps funktion is OK.

jpg63, 2018-01-29:
For the neo-8m the NMEA header is different a modification of the code of the library serialnmea.cpp #if defined (VARIOMETER_GPS_NEO6) const char rmcTag [] PROGMEM = {"GPRMC"}; const char ggaTag [] PROGMEM = {"GPGGA"}; #elif defined (VARIOMETER_GPS_NEO8) const char rmcTag [] PROGMEM = {"GNRMC"}; const char ggaTag [] PROGMEM = {"GNGGA"}; #else const char rmcTag [] PROGMEM = {"GPRMC"}; const char ggaTag [] PROGMEM = {"GPGGA"}; #endif must allow to operate the 2 GPS

damirm57, 2018-01-30:
Thank you for fast answer. I am try to integrate changes in code but obviousli in wrong way. Becose i how no exsperience in programing. When try to verify ino error message is:

C:\Program Files (x86)\Arduino\libraries\SerialNmea\SerialNmea.cpp:5:37: error: missing binary operator before token "const"

prunkdump, 2018-01-30:
Hi Damir.

Take the original code and just change the libraries/SerialNmea/SerialNmea.cpp from :
 
const char rmcTag[] PROGMEM = {"GPRMC"};
const char ggaTag[] PROGMEM = {"GPGGA"};

To 

const char rmcTag[] PROGMEM = {"GNRMC"};
const char ggaTag[] PROGMEM = {"GNGGA"};

And it's ok.

Regards,

Baptiste.

damirm57, 2018-01-30:
Awesome,  gps fix is done. Thanks Baptiste, fantastic project. I am
begginer in paragliding, thank you, thank you!!!!
Is it posibile  to conect aditional board mpu9250 without ms5611(this is
already conected) I have one more rest  from Ed tracker project.

jpg63, 2018-01-30:
I added the management of the neo-8m GPS in the branch jpg63-version on the github

the choice between neo-6 and neo-8 is done in the file variosettings.h

jean-philippe

prunkdump, 2018-01-31:
Good news !

Yes Damir you can connect your mpu9250 board on the same pins (A4 and A5).

If it don't works, add : 
-> one 4,7k resistor between A4 and 3.3V
-> one 4,7k resistor between A5 and 3.3V

Regards,

Baptiste.

damirm57, 2018-01-31:
Thank you again!

prunkdump, 2019-04-27:
It's now possible to correct this in VarioSettings.h

Thanks for the report.

closed 2019-04-27 by prunkdump