prunkdump/arduino-variometer #17: Suggestion: Beeper shouldn't beep as soon as climb rate changes

lshachar, 2018-05-27:
Hey,
I wonder if this is upsetting to anyone else except me;
whenever your climb rate changes, a new beep sound plays through the speaker, overriding the break you would normally get between two consecutive beeps in other varios.
I find it distracting, and I think of it as a bug.

Furthermore, I like the audio control you get with[ this editor tool, ](http://www.windeckfalken.de/special/xctracer/handson/main.html)
and I also like how the pitch changes smoothly while playing notes (as can be heard in the editor tool), I find it to be more informative.
I intend to rewrite some of the sound module... who knows when...

prunkdump, 2018-06-01:
Hi lshachar !

I don't notice the bug you report. Sometimes you get two consecutive beeps without break between them ? The only case where this append to me is when you enter/go out/enter the climbing zone in a very short time. This should not append often in real conditions.

But I'am not sure I correctly understand your remark.

Also, I have made a tool to setup the beeps.

-> Download Geogebra here (v5 classic)
https://www.geogebra.org

-> Open this file with geogebra
http://www.parapentiste.info/forum/index.php?action=dlattach;topic=48334.0;attach=32689

-> Change beeper.h

Regards,

Baptiste.

prunkdump, 2018-06-01:
Excuse this is the correct file : 
http://www.parapentiste.info/forum/index.php?action=dlattach;topic=48334.0;attach=32696

closed 2020-04-07 by lshachar