prunkdump/arduino-variometer #19: Cannot calibrate accelerometer (MPU6050) at all?

lshachar, 2018-07-23:
Hey,
I'm trying to use accel_calibration sketch to calibrate my vario.
I followed the directions as closely as I could, mimicking every orientation by how it's portrayed on the calibration video on youtube, and on the same order.
after taking several 'm' measurements for every orientation, when I couldn't get a lower standard deviation value, I sent 'c' to start calibration. unfortunately, the new calibration coefficients all come up as not available.

(I trimmed the information from the serial monitor, to only contain the best measurement from every orientation. I did not change the log order in any way. my notes are in **enclosure**.)

```
-------------------------
ACCELEROMETER CALIBRATION
-------------------------

Commands :
  d : display current accel vector
  m : measure accel vertor for calibration
  c : start calibration
  r : restart calibration to the beginning

Procedure :
  The accelerometer calibration procedure
  need five accel vectors from the five upper 
  orientations. That is to say with the
  accelerometer pointing :
    -> toward the sky, flat on the groud
    -> toward the left on it's left side
    -> toward the right on it's right side
    -> toward you on it's bottom side
    -> back to you, on it's top side

  For each measure :
  1) You can check the current accel vector
     with multiple 'd' command.
  2) Record the accel vector with the 'm' command
     This can be done multiple times for each orientations

  Once the five orientations recorded. You can
  launch the calibration procedure with the 'c'
  command. Use the 'd' command  to check the result.
  The distance must be as close as possible to 1.000.
  If you are not satisfied by the result. Make more
  measures or reset teh calibration with the 'r' command.
Current calibration coefficients : 
x : 0.00000
y : 0.00000
z : 0.00000

**1 FLAT ON GROUND**
-------------------------
Ready to make measure.
Don't move the accelerometer and wait...
Starting measure...
accel=(348.42666, -133.66999, 7786.48681)
Standard deviation = 51.96819
GOOD measure, recorded for calibration
-------------------------

**2 LEFT**
-------------------------
Ready to make measure.
Don't move the accelerometer and wait...
Starting measure...
accel=(-8086.21337, -105.69667, -132.12666)
Standard deviation = 54.61066
GOOD measure, recorded for calibration
-------------------------

**3 BOTTOM**
-------------------------
Ready to make measure.
Don't move the accelerometer and wait...
Starting measure...
accel=(-39.97666, -8213.40332, -266.01000)
Standard deviation = 54.84888
GOOD measure, recorded for calibration
-------------------------

**4 RIGHT**
-------------------------
Ready to make measure.
Don't move the accelerometer and wait...
Starting measure...
accel=(8298.42968, 142.45333, -505.80334)
Standard deviation = 54.18161
GOOD measure, recorded for calibration
-------------------------

**5 TOP**
-------------------------
Ready to make measure or calibration.
Don't move the accelerometer and wait...
Starting measure...
accel=(109.35334, 8180.58984, -327.39999)
Standard deviation = 54.46234
GOOD measure, recorded for calibration
-------------------------

-------------------------
CALIBRATION DONE 
-------------------------

Here the new calibration coefficients :

x : nan
y : nan
z : nan

Run multiple 'd' command to check the result.
-------------------------

**Result for sending 'd' :**
-------------------------
Ready to make measure or calibration.
Don't move the accelerometer and wait...
Starting measure...
accel=(nan, nan, nan)
Standard deviation = 56.19025
calibrated distance : nan
-------------------------
```

[image]

[image]

[image]

[image]

[image]

AvisParadocs, 2018-07-26:
+1
Same issue with mpu9250.
(#define IMU_CALIBRATION_IN_EEPROM is uncommented and static values is commented)

lshachar, 2018-07-26:
Hey AvisParadocs,
I just received an email from PrunkDump earlier today. for now, he recommends a workaround:
Do you have an SD card reader ? The problem you encounter is because I
recently changed the calibration procedure and not fully adapted the
previous sketch. If you can test :
-> connect an SD card reader
-> format an SD card with FAT16 (less than 2Go partition)
-> run the "calibration_recorder" sketch and records as many
orientations as possible ( I usually make 100 measures )
-> put the file recorded in "best-fit-calibration"
-> install Python 2
-> run the python program
-> copy the result in VarioSettings.h

(- i still haven't gotten around to try it)

prunkdump, 2018-07-26:
Hello.

I’am in vacations without computer so it’s not easy to debug the problem. 

But I have found an error for sure. Maybe someone can try if it fixe the bug.

In libraries/AccelCalibrator/accelcalibrator.cpp

Change : 

baseRadius /= (double)(3*ACCEL_CALIBRATOR_ORIENTATION_COUNT);

To

baseRadius /= (double)(ACCEL_CALIBRATOR_ORIENTATION_COUNT);

Maybe it will be fine after this.

Regards.

lshachar, 2018-07-26:
Tried it, didn't work :\
still haven't tried it with the SD card reader workaround

AvisParadocs, 2018-07-29:
Glad to hear it!
I'll prefer to use near climbing beeper instead of alarm, but in ground tests I didnt manage to see it working. At the moment trying to debug 10DOF sensor through serial port.
What was turbulence density in your test and how comfortable was vario sound in that conditions? I'm highly afraid that so high precise vario sensitivity can alarm climbing and sinking all over the place and especially in turbulence =)

lshachar, 2018-08-06:
I'm afraid calibration_recorder.ino will not build when MPU6050 or MPU6500 are defined, (will only build when MPU9150 or MPU9250 are defined)

error message:
`
calibration_recorder.ino: In function 'void makeMeasureStep()':
calibration_recorder:130: error: 'class Vertaccel' has no member named 'readRawMag'
class Vertaccel has no member named 'readRawMag'
`

lshachar, 2018-08-12:
I eventually got calibration_recorder to build (by adding  #ifdef AK89xx_SECONDARY  calibration_recorder line 129, which prevents the magnetometer readings)

and I was able to get real data from calibration_recorder. also later I installed python (you need 2.7, and install numpy library as well)

but eventually I found out that changing values in variosettings.h, such as IMU_ACCEL_CAL_BIAS, will not make any difference for the vario, since that value is not used anywhere in the program. It's one more thing that's not yet implemented.

prunkdump, 2018-08-13:
Hi !

I finally found the bug in the calibration code. Can you try again "accel_calibration" for me with the #define IMU_CALIBRATION_IN_EEPROM uncommented. And say me if it works.

@Ishachar. Good job ! For your problem, maybe you have forgotten to comment #define IMU_CALIBRATION_IN_EEPROM in VarioSettings.h. The constant IMU_ACCEL_CAL_BIAS is used in variometer.ino.

I will try to adapt my code when there is no magnetometer.

Regards,

Baptiste.

prunkdump, 2019-04-27:
I have modified the "calibration_recorder" sketch and the "calibrate.py" script to work with accelerometer without magnetometer.

Thanks for the report.

closed 2019-04-27 by prunkdump