prunkdump/arduino-variometer #23: #define MPU6050 causes a compiler error

Lion1971, 2018-09-13:
Hello,
if I use a MPU6050 by #define MPU6050 in the variosettings the compiler for the variometer.ino causes the following compiler error:

"C:\...\Arduino\libraries\vertaccel\vertaccel.cpp:333:1: error: expected '}' at end of input
 }
 ^
exit status 1"

If I define  a MPU9250 (which I don't have) it can be compiled.

In the old version of the JPG63 didn't have that issue.

Thank you and best regards
Lion

Lion1971, 2018-09-15:
Hello,
here is the result of another test. I changed in the variosettings.h the calibration method from "by static value" to "by EEPROM". Then the compiler message is the following.
Best regards
Lion

C:\...\arduino-variometer-jpg63-version\arduino-variometer-jpg63-version\variometer\variometer.ino:388:9: warning: extra tokens at end of #endif directive [-Wendif-labels]

  #endif HAVE_SCREEN_JPG63
         ^
C:\Users\...\Documents\Arduino\libraries\vertaccel\vertaccel.cpp: In member function 'uint8_t Vertaccel::dataReady()':

C:\Users\...\Documents\Arduino\libraries\vertaccel\vertaccel.cpp:222:33: error: qualified-id in declaration before '(' token

 uint8_t Vertaccel::dataFullReady(double* accel, double* quat, double* v, double* n) {
                                 ^
C:\Users\...\Documents\Arduino\libraries\vertaccel\vertaccel.cpp:333:1: error: expected '}' at end of input
 }
 ^

exit status 1

prunkdump, 2018-09-15:
Ok I'have updated the code.

Try if it's good now.

Lion1971, 2018-09-15:
Thank you Baptiste for your work :-)

The variometer.ino can be compiled with the following warning (see below), but is uploaded.
The only issue is now, that the vertical speed is around -1m/s also after a new calibartion. I tried calibration method "by EEPROM" and "by static". Before it was around 0m/s

Best regards
Lion

Compiler warnings:

#endif HAVE_SCREEN_JPG63
         ^
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp: In member function 'void ScreenScheduler::displayStep()':
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp:734:24: warning: passing 'const VarioScreen' as 'this' argument discards qualifiers [-fpermissive]
   if( screen.clearStep() ) {
                        ^
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp:224:6: note:   in call to 'bool VarioScreen::clearStep()'
 bool VarioScreen::clearStep() {
      ^
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp: In member function 'void ScreenScheduler::setPage(int8_t)':
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp:773:21: warning: passing 'const VarioScreen' as 'this' argument discards qualifiers [-fpermissive]
   screen.beginClear();
                     ^
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp:220:6: note:   in call to 'void VarioScreen::beginClear()'
 void VarioScreen::beginClear() {
      ^

Lion1971, 2018-09-15:
Sorry I pushed the wrong buttom (Close).

prunkdump, 2018-09-16:
Hi Lion.

You see -1.0m/s on the screen ?! Or just -0.1m/s ?

First to be sure.
-> Move all files in your "Arduino" folder in some backup place.
-> Download the latest "master" branch code 
https://github.com/prunkdump/arduino-variometer/archive/master.zip
-> copy the content of the arduino-variometer-master folder inside your arduino folder
-> Change only in VarioSettings.h : the MPU and your pins 

Compile and upload. And check the result. Say me what you se in the screen.

Regards,

Baptiste.

Lion1971, 2018-09-16:
Hi Baptiste,

I downloaded the master like you suggested instead of the JPG63 and the result is the same, but no compiler warnings.
[image]
 When the vario is laying on the table the value differs between -1,0m/s and -1,1m/s.

Regards
Lion

prunkdump, 2018-09-16:
Ok this is a bug. Not a calibration problem.

Can you enable the « calibration in EEPROM » and run the accel_calibration sketch ?

Or the calibration_recorder sketch if you have a working SD card.

I just need one raw measure of your accelerometer. 

If you run the accel_calibration sketch send me one measure output.

If you run the calibration_recorder sketch send me the file recoded on the SD card.

Regards,

Baptiste.

prunkdump, 2018-09-17:
Thanks.

Last question. When you upload the master version with all the default parameters.
-> You have -1.0m/s when the variometer is flat on the table

But what do you have if you put the screen toward the ground ?
-> -1.0m/s
or 
-> 1.0 m/s 
??

Can you also make the full calibration procedure with "accel_calibration" and send to me the full serial output.

Regards,

Baptiste.

Lion1971, 2018-09-17:
Before I send a wrong input.

From the masters default settings I changed the MPU from 9250 to 6050 or should I try 9250 even if I have the 6050 HW?

The full calibration should be done with the "m" command at the seriel console?

Regards
Lion

prunkdump, 2018-09-17:
Excuse me.

Yes you set mpu6050.

And for the full calibration : 
-> you set calibration in eeprom
-> you upload "accel_calibration"
-> you make an "m" command for each orientation
-> you start calibration with the "c" command
-> you send to me the full console Serial Log.

Regards,

Baptiste.

Lion1971, 2018-09-17:
Here are the answer and result of the tests I did.

Flat on the table it is -1,0m/s
If I put the screen toward the ground it shows +1,3m/s.

Result of the measures (2 values per orientation) and the calibration. I deleted some redundant lines with "Ready to ..."

-------------------------
Ready to make measure.
Don't move the accelerometer and wait...
Starting measure...
accel=(906.69000, -682.41998, 7141.47656)
Standard deviation = 81.15827
GOOD measure, recorded for calibration
-------------------------
Ready to make measure or calibration.
-------------------------
Ready to make measure or calibration.
Don't move the accelerometer and wait...
Starting measure...
accel=(872.59997, -730.93664, 7140.11328)
Standard deviation = 74.12198
GOOD measure, recorded for calibration
-------------------------
Ready to make measure or calibration.
-------------------------
Ready to make measure or calibration.
Don't move the accelerometer and wait...
Starting measure...
accel=(40.52667, 8027.25000, -181.44000)
Standard deviation = 71.99639
GOOD measure, recorded for calibration
-------------------------
Ready to make measure or calibration.
-------------------------
Ready to make measure or calibration.
Don't move the accelerometer and wait...
Starting measure...
accel=(42.70333, 8024.37988, -140.61666)
Standard deviation = 74.07876
BAD measure, not recorded, possible problems are :
-> ambiguous orientation
-> orientation already done with lower standard deviation
-------------------------
Ready to make measure or calibration.
-------------------------
Ready to make measure or calibration.
Don't move the accelerometer and wait...
Starting measure...
accel=(8578.80371, -250.69332, -1371.96667)
Standard deviation = 78.79714
GOOD measure, recorded for calibration
-------------------------
Ready to make measure or calibration.
-------------------------
Ready to make measure or calibration.
Don't move the accelerometer and wait...
Starting measure...
accel=(8580.21679, -253.15666, -1350.05004)
Standard deviation = 72.09682
GOOD measure, recorded for calibration
-------------------------
Ready to make measure or calibration.
-------------------------
Ready to make measure or calibration.
Don't move the accelerometer and wait...
Starting measure...
accel=(699.50335, -8376.92968, -1594.63000)
Standard deviation = 66.59497
GOOD measure, recorded for calibration
-------------------------
Ready to make measure or calibration.
-------------------------
Ready to make measure or calibration.
Don't move the accelerometer and wait...
Starting measure...
accel=(707.22332, -8380.69042, -1473.22338)
Standard deviation = 62.68950
GOOD measure, recorded for calibration
-------------------------
Ready to make measure or calibration.
-------------------------
Ready to make measure or calibration.
Don't move the accelerometer and wait...
Starting measure...
accel=(-7836.50976, -342.60998, -730.08666)
Standard deviation = 67.58684
GOOD measure, recorded for calibration
-------------------------
Ready to make measure or calibration.
-------------------------
Ready to make measure or calibration.
Don't move the accelerometer and wait...
Starting measure...
accel=(-7840.72998, -62.40000, -773.97668)
Standard deviation = 64.91309
GOOD measure, recorded for calibration
-------------------------
Ready to make measure or calibration.
-------------------------
CALIBRATION DONE 
-------------------------

Here the new calibration coefficients :

x : 354.71286
y : -167.63986
z : -1055.46362

Run multiple 'd' command to check the result.
-------------------------
-------------------------
Ready to make measure or calibration.
------------------------- 

Best regards
Lion

prunkdump, 2018-09-17:
Thanks !

Comment now the calibration in EEPROM en VarioSettings.h. And set this static value : 

#define IMU_ACCEL_CAL_BIAS {23696, -10224, -67324}

Say me if this correct your problem before investigating further.

Baptiste.

Lion1971, 2018-09-17:
The value is the same -1,0m/s

The compile shows the following warning:

D:\...\arduino-variometer-master\arduino-variometer-master\variometer\variometer.ino:61:1: warning: narrowing conversion of '-67324l' from 'long int' to 'int16_t {aka int}' inside { } [-Wnarrowing]
 };
 ^
D:\...\arduino-variometer-master\arduino-variometer-master\variometer\variometer.ino:61:1: warning: overflow in implicit constant conversion [-Woverflow]

Regards Lion

prunkdump, 2018-09-17:
So I understand the problem !

Your mpu need a very high calibration coefficient. And this overflow the 16bit variable containing it.

Are you sure there is not something that apply pressure on you mpu chip ? It give very faulty values.

But I have a workaround method.
In libraries/vertaccel/vertaccel.h change :

#define VERTACCEL_ACCEL_CAL_BIAS_MULTIPLIER 4

And in variosettings. Divide all the bias coefficient by 4 ( take the nearest integer ).

Lion1971, 2018-09-17:
Thank You Bapiste for the great support, the workaround is fine and now it is around 0m/s

Because I have two devices and both worked with the old version of the JPG63 I have some questions:

1. What was different in the old version and are there advantages in the new version that I should update? 

2. Because two independent MPU6050 devices have the problem shouldn't this workaround be the standard in the new version?

3. Can I do the accel_calibration_nointeractive and use the EEPROM also with the workaround?

4. When I like to use the new JPG63 version I have to transfer the workaround from the master to the JPG63 version, right?

Regards Lion

Lion1971, 2018-09-18:
Thank you I will test and send the result this evening.

For 1) Do I have an advantage with magnetometer? Means a new feature now or in the future?

Which MPU type is your recommendation 9050, 9250...? They are not expensive to exchange.

Regards Lion

prunkdump, 2018-09-18:
Actually the magnetometer is functional but not used. It's for the future (compas, compute wind speed ...)

My recommendation is MPU9250. You can find it already coupled with the ms5611.

Thanks for the tests.

Regards.

Baptiste.

Lion1971, 2018-09-18:
Hi!
the workaround works also with the new JPG63 version with accel_calibration_nointeractive and usage of the EEPROM. Thank you so much :+1: 

The only issue what appears again during the compilation are the following warnings regarding the varioscreen. I don't know if it is a serious problem or it can be irgnored? The screen shows normal values.

Best regards
Lion

D:\...\arduino-variometer-jpg63-version\arduino-variometer-jpg63-version\variometer\variometer.ino:388:9: warning: extra tokens at end of #endif directive [-Wendif-labels]

  #endif HAVE_SCREEN_JPG63
         ^
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp: In member function 'void ScreenScheduler::displayStep()':
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp:734:24: warning: passing 'const VarioScreen' as 'this' argument discards qualifiers [-fpermissive]
   if( screen.clearStep() ) {
                        ^
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp:224:6: note:   in call to 'bool VarioScreen::clearStep()'
 bool VarioScreen::clearStep() {
      ^
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp: In member function 'void ScreenScheduler::setPage(int8_t)':
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp:773:21: warning: passing 'const VarioScreen' as 'this' argument discards qualifiers [-fpermissive]
   screen.beginClear();
                     ^
C:\Users\...\Documents\Arduino\libraries\varioscreen\varioscreen.cpp:220:6: note:   in call to 'void VarioScreen::beginClear()'
 void VarioScreen::beginCl_ear() {
      ^

prunkdump, 2018-09-18:
Perfect !

Thanks you for investigating the problem.

I close the issue. If you want to contact me. You can use my email on my profile.

Regards,

Baptiste.

closed 2018-09-18 by prunkdump