prunkdump/arduino-variometer #26: Issue after HW upgrade to NEO6MV2 and 10DOF (5611/9250)

LL03, 2018-11-28:
Hello,
I replaced my old GPS module by the blue NEO6MV2 and the 10DOF sensor whith combined MS5611 and MPU9250. Before it was the red NEO6M and the MS5611 and MPU9250 separate. 
I am using the jpg63 software. With the new software the display starts blinking after around 30s and is frozen and no sound. Then it restarts and it could also happen that the display is mixed up and there is a continous sound before it restarts again.

The display is OK
[image]
The display is mixed up and the attitude changes to baro value. After the automatic restart it is like the first picture again.
[image]

I read the issue "Using a NEO6MV2 (XM37-1612) , old GPS" and changed the NMEA_Parser settings and the restart problem is solved but the shown valeus for the time and attitude are wrong.

//#define NMEA_PARSER_RMC_SPEED_PRECISION 1000.0
//#define NMEA_PARSER_GGA_TIME_PRECISION 100
//#define NMEA_PARSER_GGA_ALTITUDE_PRECISION 10.0
#define NMEA_PARSER_RMC_SPEED_PRECISION 100 // initialement 1000.0
#define NMEA_PARSER_GGA_TIME_PRECISION 1000   //initialement  100
#define NMEA_PARSER_GGA_ALTITUDE_PRECISION 100.0  //initialement  10.0

What is the meaning of the parameters and what are valid values. Which can I use that the values are correct but my restart problem is further solved?

Best regards
LL03

LL03, 2018-11-28:
So I did another test an removed the TX line from the GPS. It remains the same issue, that I now think it is the 10DOF sensor whith combined MS5611 and MPU9250 (purple version). Could it happen if the 10DOF is not delivering correct or constantly values? I build two devices and the behaviour is the same.

prunkdump, 2018-11-28:
Hi LL03 !

The problem you describe seems to be a powering problem.

When you power-on your variometer something need to much current that you power circuit can't feed. So the voltage decrease greatly and the ms5611 ( very sensitive to power ) give bad values. This is because your screen display too big numbers.

Sometimes the voltage decrease so much that is goes under the Arduino brown out detection. So a reset is launched and the Arduino reboot.

How do you power your components ? Have you changed some voltage regulators ?

The "blue" neo-6M does not not need code modification. You can keep the original code. But it need power as it is a little hungry ;)

Regards,

Baptiste.

LL03, 2018-11-28:
Hi Baptiste,
I will measure the overall current. I have a Lipo as power supply all componeents are directly connected to the Lipo except the Display and MicroSD card reader, but without SD card in it. They are connected to VCC (3.3V). I will desolder the reader.
Thank you I will inform you.
Regards
LL03

LL03, 2018-11-28:
Hi Baptiste,
I measured the current an it is around 100mA and the card reader alone without SD Card only some mA. But without the card reader the vario works. I assume that the connected signal pins are the root cause.
Do you recommend a specific micro SD card reader and can I use a 3.3V type?
I think the issue can be closed.
Regards
LL03

LL03, 2018-11-28:
Hi Baptiste,
the Lipo has 1000mA and the voltage at the supply pins is 3,8V. I desoldered the reader an now it works. Then I used a different microSD reader type, also with 3,3V and it works too without reset. Now it can also write IGC files. So I will replace the modules. For the support two pictures. The 1st which works fine and the 2nd which is not usable.
[image]
[image]
Regards
LL03

prunkdump, 2018-11-29:
Thanks for your feedback !

But I think your previous SD card module have a problem. Some short circuit maybe.

i use the same module than you : 

https://prunkdump.github.io/GNUVario/tutorials/boards.html

Regards,

Baptiste.

prunkdump, 2018-11-29:
I close the issue.

closed 2018-11-29 by prunkdump

LL03, 2018-12-03:
Many thanks for the support:-)
Can I ask a question or should I open a new topic?
With the other module (which you use) I have the reset problem sometimes too without SD  Card in the slot. 
When I solder a SD module and configure "Have SD card", is it mandatory to plug in a SD card?
If this configuration causes problems, can I solve it by a configuration change: Deactivate "Have SD card"?
If that is the case can the programm be more tolerant if an SD card is not available?
Best regards
LL03

prunkdump, 2018-12-17:
Someone give the solution to me !

Simply put a wire between the IMU AD0 pin and ground (GND) !

I think this will solve your issue !

Regards,

Baptiste