prunkdump/arduino-variometer #27: Missing LICENSE

TechnologyClassroom, 2019-02-08:
I was linked to this page from https://prunkdump.github.io/GNUVario/.

There is no license for this project. The default is copyright.

You should license this project under the AGPL3 or GPL3 license. If you would like me to submit a PR to make this easier, I would be happy to do that.

prunkdump, 2019-02-23:
Hi !

Excuse me for the delay, I'm working hard on the next release where the whole code is redesigned.

Don't worry, the code will be updated soon to fit the GNU GPLv3 licence.

If you need some part of the code for another project don't hesitate. I will give copyleft on the next release.

Regards,

Baptiste.

prunkdump, 2019-03-14:
Hi Michael !

I have released the whole project under GPLv3 now.

Thanks for the help doing this !

closed 2019-03-14 by prunkdump