prunkdump/arduino-variometer #31: Automatic change between summer and winter time

Lion1971, 2019-04-22:
For paraglider who would like to use the variometer mainly for flying for a long time, it would be helpful, if the summer and winter time will be changed automatically end of March and end of October. 

For non technicans the vario could be used the complete season without changing the variosettings.h and uploading of the variometer code every half year.

JohnRobson, 2019-10-26:
Maybe the pilot will want to fly more hours in the summer, but how the summer and winter time affect the vario behavior?!

JohnRobson, 2019-10-26:
@Lion1971  Maybe the pilot will want to fly more hours in the summer, but how the summer and winter time affect the vario behavior?!

lshachar, 2020-04-07:
@JohnRobson , It would affect the timestamp created in the IGC tracklog files.
@Lion1971, I see that VARIOMETER_TIME_ZONE is only used in the code once, so it can be exported to a file saved on the SD card, which gets read when you start your vario. I think it's a little easier than having to recompiling and reuploading every six months. (However I don't think it should be automatically changed in any set date, because not all countries change daylight savings time on set dates!)