prunkdump/arduino-variometer #41: user manual

alkovar, 2020-10-12:
Hi,
a friend of mine brought to my attention this awesome project and I just completed my first vario yesterday. It does work! But I struggle to find any relevant information about bluetooth pairing, meanings of the symbols on the screen etc. Is there a quick start/user manual, which would explain how to use the vario once it is put together? The VarioSettings.h has some comments but they don't explain everything...

lshachar, 2021-08-21:
Can't help you much, but for what it's worth, it took me over two years with this vario to figure out that a quick tap on it gets it to go into mute mode, and another tap gets it back to normal beeping mode. My first prunkdump vario was headless (without a screen) and there was really no way, no documentation, to read that this was a feature. I thought my arduino board was buggy. On my second vario I have a screen, and it still took me such a long time to finally figure out that it's the tapping (or sometimes, can be triggered by a very strong sink/thermal) that caused the mute mode to trigger.

Hasi123, 2021-08-22:
There is no manual besides https://prunkdump.github.io/GNUVario/
@lshachar I disabled the mute on tap feature as soon as I started to use it, because it quite often triggered itself.

lshachar, 2021-08-22:
@Hasi123 If only I figured by time that it was indeed a feature, and not a bug!
I think this feature should be noted / configured in the varioSettings.h file.
(Or it might have already been, since I'm running an old version of the code)
If you care to write how you disabled it, might help someone else in the future.

Hasi123, 2021-08-22:
> @Hasi123 If only I figured by time that it was indeed a feature, and not a bug!
> I think this feature should be noted / configured in the varioSettings.h file.
> (Or it might have already been, since [...]

Here is the commit on my GitHub: https://github.com/Hasi123/HaskoVario/commit/66532e42839baea272d0faa46b3a1db0d5d9cde7